toitware/public #5: The free account has zip. archive issues

surayyaurazimbetova, 2021-04-11:
When downloading archived files as depicted in below image: lib.zip and example.zip in zip format, it is not straightforward how to extract those on windows. 
[image]

Also it would be nice if it was possible to deploy a "hello world" program on the virtual emulator

HelenaMeyer, 2021-04-13:
Thanks. 
The lib.zip and examples.zip can be extracted on Windows using 7zip (https://www.7-zip.org/).

Alternatively you can use the Toit CLI to automatically download and extract the two, by running the following commands:

`toit sdk examples ~/Desktop/toit-examples`

to extract Toit example code into a new folder, here called `toit-examples`, and

`toit sdk lib ~/Desktop/toit-lib`

to extract the Toit library into a new folder, here called `toit-lib`

HelenaMeyer, 2021-04-13:
You can deploy an app on a simulator with the Toit CLI like this. 
First, download the examples, then create the simulator, and deploy an app:

```
$ toit sdk examples toit-examples
5.97 KiB / 5.97 KiB [-------------------------------------------------------------------------------------------------------------------------------------------] 100.00% ? p/s
Downloaded Toit SDK examples to /home/user/toit-examples for SDK: v1.0.2

$ toit simulator start --alias my-simulator
Simulator started:
	Simulator ID: d1f422fd-ea29-4e78-8ba4-995097ed04f6
	Device ID: 71ed2ad6-e845-4925-87a0-cdc6b922f503
	Device Name: my-simulator
	Firmware: v1.0.2

$ toit deploy -d my-simulator toit-examples/examples_v1.0.2/hello/hello.yaml 
successfully deployed the job HelloApp on my-simulator

```
You can see the app installed on your device in the Apps column for the device in https://console.toit.io/

floitsch, 2021-04-13:
As a follow-up: the reason you need is 7-zip, is because we accidentally created a tar.gz instead of a zip file. The file (`examples.zip`) isn't actually a zip file.

We will fix this.

jstroem, 2021-04-14:
This has been fixed and will be effective in the next SDK release. Until then the CLI cmd `toit sdk examples` can be used to get examples.

HelenaMeyer, 2021-06-11:
This has been fixed and can be closed now

closed 2021-06-11 by HelenaMeyer