toitware/public #6: Allow user-defined objects to be encoded and decoded with JSON library

floitsch, 2021-05-02:
Currently the JSON library only supports encoding and decoding of several hard-coded types:
```
  encode obj:
    if obj is string: encode_string_ obj
    else if obj is num: encode_number_ obj
    else if identical obj true: encode_true_
    else if identical obj false: encode_false_
    else if identical obj null: encode_null_
    else if obj is Map: encode_map_ obj
    else if obj is List: encode_list_ obj
    else: throw "INVALID_JSON_OBJECT"
```
It would be nice to allow user-defined objects as well, and maybe other objects (like `Duration` and `Time`).

See https://stackoverflow.com/questions/67355048/serializing-deserializing-objects-to-from-json-on-toit-language

Molbeck, 2021-11-22:
I agree with @floitsch. This is something which is needed - at least for me as well.

floitsch, 2022-09-10:
Encoding is supported now (and has been for a while).
The encoder takes a block that is invoked for objects that don't map directly to JSON.
https://libs.toit.io/encoding/json/library-summary#encode(2%2C1%2C0%2C)