toitware/public #7: Currently impossible to listen on a port

floitsch, 2021-06-08:
Currently the following doesn't work.
```
import net

main :
  network_interface := net.open
  log("network_interface -- ok")
  serverSocket := network_interface.tcp_listen 8080
  serverSocket.listen "localhost"
  log ("wait...")
  localAddr := serverSocket.local_address
  socket := serverSocket.accept
  log ("Accept was done...")
```

The following code has been found to be the culprit:
```
 if (!address->slow_equals("localhost") && !address->slow_equals("127.0.0.1")) {
      // We currently only implement binding to localhost or INADDR_ANY.
      UNIMPLEMENTED_PRIMITIVE;
    }
```
We need to allow more addresses here.

kasperl, 2021-06-24:
We have made progress on this in SDK v1.1.0, but we have found an additional issue.

kasperl, 2021-06-28:
We have just shipped SDK v1.1.1 which solves the last known issue with port listening.

kasperl, 2021-06-29:
The solution provided in SDK v1.1.1 has been verified and listening now works.

closed 2021-06-29 by kasperl