toitware/public #9: Bluetooth Support within Toit

juergenspecht, 2021-06-14:
**Is your feature request related to a problem? Please describe.**
I am using ESP32's as kinda a gateway between "local" (as in: physically close to the ESP32) Bluetooth devices and the Internet, which allows me to gather data from local sensors and make them available for example on a Website or within a Home-IOT System. If Toit had Bluetooth functionality, it would be quite simple to build Bluetooth/Web interfaces (especially since the integrated Pub/Sub functionality is already there).

**Describe the solution you'd like**
Currently I mostly use NimBLE, because it does all I need and uses less memory than other solutions. However, I would not describe it as "elegant" or "straightforward". So I would like to know less about the inner workings of Bluetooth and just use a simple abstraction within Toit, the language.

**Additional context**
Since mostly I scan "passive" to save on battery power on the Bluetooth devices and some of the devices (for example the Xiaomi Mijia Bluetooth Temperature Humidity Sensor comes to mind) are quite chatty even in passive mode, I could live with a staggered release of this feature. For example release "scan and read" functionality first and the more advanced features ("pairing, write, etc") later.

john7doe, 2021-06-17:
I would also love to see Bluetooth (Actually just BLE) support within Toit. My use-case requires the device to be a GAP Peripheral/GATT server exposing HID and Battery service.

kasperl, 2021-10-08:
Fixed in v1.4.0 :)

See https://docs.toit.io/platform/tutorials/ble/advertising as an example.

closed 2021-10-08 by kasperl