toitware/public #10: [BUG] Fleet prefix not working on toit command line tool

marianopeck, 2021-07-16:
**Describe the bug**
Hi, 
I am trying to prefix the devices of my fleet following the instructions from [here](https://docs.toit.io/platform/console/fleet) but it doesn't work. 

**To reproduce**
Steps to reproduce the behavior:
1. Execute `toit fleet -p "RIoT"`
2. No renaming is happening and on the console I see all the help...like if the command wouldn't have been valid. 

**Expected behavior**
The devices should have been prefixed. 

**Screenshots**
[image]

**SDK and console version (please complete the following information):**
Run `toit version`
Toit CLI:
+--------+------------+
| v1.7.0 | 2021-06-24 |
+--------+------------+

Thanks in advance,

floitsch, 2021-07-16:
The fleet command is designed to execute a specific operation on multiple devices at the same time.
The available operations are `config`, `deploy`, ...

As such, you could write: `toit fleet firmware-status` to get the firmware-status of all your fleet.

Frequently, subsets of all devices (your fleet) are handled differently. For example, you might have some devices in a room close to you, while others are harder to reach. If you want to execute an operation only on a subset of devices you can use the `--name-prefix` flag. It executes the operation only on the devices that have the specific prefix.

For example, the command `toit fleet firmware-status --name-prefix="florian-"` only runs `firmware-status` on devices that have a name that starts with "florian-":
```
ᐅ build/toit fleet firmware-status --name-prefix="florian-"
   ◦ Device 'florian-original-act-cam', firmware status: v1.1.2 installed
   ◦ Device 'florian-laughing-economy', firmware status: v1.2.0-pre.70+471586f23 installed
   ◦ Device 'florian-precious-trash2', firmware status: v1.0.2 installed
   ◦ Device 'florian-colorful-drama', firmware status: v1.1.0-pre.111+c77be5cc1 installed
```

This also explains why you get an error message: `toit fleet -p "RIoT"` is not telling the `toit fleet` command which operation you want to execute. Translated to English it says: "on all my devices that start with RIoT do ..." without saying what should actually be done. 

If you just want to rename all devices so they are prefixed with "RIoT-" you could try the following bash-script:
```
for name in $(toit devices -o csv | tail -n +2| cut -d',' -f2); do toit device rename "RIoT-$name" -d $name; done
```
As a dry-run add an `echo` to see which commands the script would execute:
```
for name in $(toit devices -o csv | tail -n +2| cut -d',' -f2); do echo toit device rename "RIoT-$name" -d $name; done
```

marianopeck, 2021-07-16:
Hi @floitsch 

Thanks a lot for the detailed and quick reply. Now I understand...I just got confused by the documentation. 

Have a nice weekend,

closed 2021-07-16 by marianopeck