toitware/public #12: [BUG] Infinitive loop when trying to upgrade firmware?

marianopeck, 2021-07-17:
**Describe the bug**
I am trying to upgrade v1.0.3 -> v1.1.3. However, it's been more than 60 minutes and the process seems to keep running. When I see the Troubleshoot section I do see kind of an infinitive loop, but I am not sure. I don't know either how long it normally takes as this is the first time I upgrade the firmware....but 60 minutes tells me there is something wrong.

I tried to copy the contents of the Troubleshoot to share with you but I couldn't find any easy way to do that. I am missing something?

I took some screenshots at least... it starts with this:

[image]

Then you can see how it starts again:

[image]

**SDK and console version (please complete the following information):**
Run `toit version`
Toit CLI:
+--------+------------+
| v1.7.0 | 2021-06-24 |
+--------+------------+

marianopeck, 2021-07-17:
Funny enough....it seems to have finished correctly after 90 minutes. So..maybe I was just impatient...

lask, 2021-07-19:
@marianopeck, glad to here that you managed to update the device. It is not supposed to take 90 minutes to update the device, so we will investigate this issue.

FYI the second troubleshooting screenshot contains your wifi credentials.

marianopeck, 2021-07-19:
Thanks @lask 
BTW, I don't mind sharing you my credentials or IDs (organization and hardware ID?) if that helps you debug the problem.  Just let me know privately. 
Best,