toitware/public #13: Be possible to un-provision or force a provision?

marianopeck, 2021-08-09:
**Is your feature request related to a problem? Please describe.**
I have an ESP32 that I use to demo the whole process of getting started with Toit (and that involves the first step of provision). I also like testing the provision in different operating systems, in Chrome with the web console, etc. Some time ago, I *think* I was able to re-provision a device. But now it seems I can't.  And of course, while they are cheap, I don't want to buy a new one each time I want to run a provision. 

**Describe the solution you'd like**
I would like the `provision` command to allow a `-f force` kind or argument or a new command `unprovision`. Whatever that allows me to provision it again just for demoing or testing purposes. 

Thanks in advance,

jstroem, 2021-08-16:
This is intentional from our side.
One physical device should only have one hardware id.
If you really want to do this, you can use the `esptool` to erase the flash for the esp32. That way we can't distinguish your device from a new one:
```
esptool erase_flash
```
or using the cached esptool installed by toit:
```
~/.cache/toit/tools/esptool_linux_v3.0 erase_flash
```

closed 2021-08-16 by jstroem

marianopeck, 2021-08-20:
Thanks @jstroem 
I just tried on OSX and Windows and in both platforms it worked perfectly. I was able to erase the flash, unclaim and provision again. 
Thanks!