toitware/public #22: [BUG] Can't create a metric subscription

marianopeck, 2021-09-20:
**Describe the bug**
I am trying to create a metric subscription called "temperature" through the web console and I am getting an error. 

**To reproduce**
Steps to reproduce the behavior:
1. Go to the web console, `Data` tab, `create subscription` button, select `metrics` type and enter `temperature` as the name.
2. See error

**Screenshots**

[image]

**Additional context**
I also tried to do it from the CLI with `toit data subscription create metrics temperature` (as it is in the HELP) but I get a message saying "the metrics data type has not yet been created"

jstroem, 2021-09-23:
Hi @marianopeck 

First and foremost, thanks for this very through bug-report. 

Just to give some background on why this happens:
No metrics has ever been made for a device in your project, so the underlying topic is has not been created yet and therefore the subscription cannot be created.

The same logic is used for data subscriptions of other types so I will look into fixing it once and for all.

That said, we are in the process of removing the metrics libraries from Toit since they in their current form are not that useful compared to the general pubsub library.

jstroem, 2021-09-24:
Fix has been rolled out.

closed 2021-09-24 by jstroem