toitware/public #27: MQTT client subscribe crashes after some 30 seconds

xal88, 2021-10-17:
**Describe the bug**
Can subscribe to a mqtt_topic and read some values. Crashes after some 30 seconds every time.
No other app is running in this ESP32. Tried on 2 different ESPs.

**To reproduce**

* Code from git example
```
import mqtt
import net

main:
  socket := net.open.tcp_connect "172.21.14.253" 1883
  client := mqtt.Client
    "toit-client-id"
    mqtt.TcpTransport socket

  task::
    client.handle: | topic/string payload/ByteArray |
      print "$Time.now.utc Received message on topic '$topic': $payload"

  client.subscribe "device/events" --qos=1

  client.publish "device/data" "Hello World".to_byte_array
```
* log result:
```
[mqtt_example.toit] 2021-10-16T11:20:37.533479Z: <process initiated>
[mqtt_example.toit] 2021-10-16T11:20:37 Received message on topic 'device/events': #[0x32]
[mqtt_example.toit] 2021-10-16T11:21:29 Received message on topic 'device/events': #[0x31, 0x39]
[mqtt_example.toit] 2021-10-16T11:22:01 Received message on topic 'device/events': #[0x31, 0x31]
[mqtt_example.toit] 2021-10-16T11:22:08.469751Z: <stack trace>
[mqtt_example.toit] EXCEPTION error. 
[mqtt_example.toit] UNEXPECTED_END_OF_READER
[mqtt_example.toit]   0: BufferedReader.ensure.<block> <sdk>/reader.toit:92:16
[mqtt_example.toit]   1: BufferedReader.ensure_    <sdk>/reader.toit:61:27
[mqtt_example.toit]   2: BufferedReader.ensure     <sdk>/reader.toit:92:5
[mqtt_example.toit]   3: BufferedReader.byte       <sdk>/reader.toit:151:5
[mqtt_example.toit]   4: BufferedReader.read_byte  <sdk>/reader.toit:426:10
[mqtt_example.toit]   5: Packet.deserialize        <pkg:mqtt>/packets.toit:21:21
[mqtt_example.toit]   6: TcpTransport.receive      <pkg:mqtt>/tcp.toit:31:18
[mqtt_example.toit]   7: Client.run_               <pkg:mqtt>/client.toit:145:28
[mqtt_example.toit]   8: Client.<lambda>.<block>.<block> <pkg:mqtt>/client.toit:52:11
[mqtt_example.toit]   9: catch.<block>             <sdk>/core/exceptions.toit:111:10
[mqtt_example.toit]  10: catch                     <sdk>/core/exceptions.toit:109:1
[mqtt_example.toit]  11: catch                     <sdk>/core/exceptions.toit:60:10
[mqtt_example.toit]  12: Client.<lambda>.<block>   <pkg:mqtt>/client.toit:51:9
[mqtt_example.toit]  13: Client.<lambda>           <pkg:mqtt>/client.toit:49:30
[mqtt_example.toit] 
[mqtt_example.toit] 
[mqtt_example.toit] 
[mqtt_example.toit] 2021-10-16T11:22:08.519444Z: <process terminated - exit code: 0>
```

**Expected behavior**
working stable for hours...

**SDK and console version (please complete the following information):**
FW 1.4.1 
MQTT client 1.1.1
+---------+------------+
| v1.12.0 | 2021-10-08 |
+---------+------------+

andersjohnsen, 2021-10-17:
I can mirror this behavior locally. I suspect a missing ping/pong handler. I'll go ahead and add that.

andersjohnsen, 2021-10-17:
Fixed and released as mqtt v1.1.2.

closed 2021-10-17 by andersjohnsen

andersjohnsen, 2021-10-17:
Limited information here, but it looks like the device lost Wifi connectivity. Let me know if it continues to behave like this.

xal88, 2021-10-17:
deploy onto a second ESP32 very close to the AP with 100% signal, running for 2h now w/o reboots.
First ESP has only 20% network signal.
Stopping the WLAN crashes both.  Will report longterm test...