toitware/public #31: Allow Device End-Users to Reconfigure Wifi Connection

keenanjohnson, 2021-11-18:
**Is your feature request related to a problem? Please describe.**
One of the critical workflows for any IoT product is allowing device end-users to connect their device to their home wifi or change their wifi if they move. I would like to request this feature in Toit, which allows Toit to be used by product developers everywhere! I have been told in the [Toit Slack](https://toit-chat.slack.com/archives/C017YFJT8RM/p1637200043050700) that the Toit API would need to be updated to allow this change.

**Describe the solution you'd like**
The Toit API should be updated to allow configuration changes of the device Wifi Network. This would then allow someone to reconfigure the wifi connection details via a webpage hosted by the esp32 or via a Bluetooth connection using an app.

See these examples of IoT devices and the workflow they use to configure the Wifi connection. Toit needs to allow something similar:

- https://www.bhencke.com/pixelblazegettingstarted
- https://github.com/balena-os/wifi-connect

**Describe alternatives you've considered**
The alternative would be to allow end device users to have a login to the Toit Web Service and provide them with only access to their device's information, though this seems more complex than exposing the info on the device as described above.

**Additional context**
Add any other context or screenshots about the feature request here.

Sakvojage, 2022-02-18:
That would be the gateway to using devices commercialy. 
In my case - sending WIFI settings via serial would be the best way, so I could incorporate this into website.

thisisnotcamilo, 2022-04-28:
Is there any news on this feature? It will be very useful for us trying to deliver commercial products using the toit platform.