toitware/public #32: [BUG] No way to add a claimed device when no device exists yet

floitsch, 2021-11-24:
When a user doesn't have any device the "claim device" button isn't visible.

Current work-around is to add a simulator first.

jstroem, 2021-12-01:
Go to Serial, connect the device and use the "Claim" button from there.

floitsch, 2021-12-02:
Agreed that it's a minor issue, but separate claiming is often when you don't have the device connected to the machine.
For example I had one user where the claiming initially failed because there was a brownout on the USB port.
As such, it's not really easy to connect the device since I had to move it away from the computer in the first place. Fwiw, it ended up being easier to just run `toit device claim ...` anyway, but still.