toitware/public #35: [cli] Allow device (and other settings) to be specified using environment variables

ptxmac, 2021-11-28:
**Is your feature request related to a problem? Please describe.**
When working on multiple projects, I often have different hardware setups for each, and deployment needs to be targeted to a specific device.
Every time I switch I have to use `toit device use <device>` (or add `--device` to deploy). 

**Describe the solution you'd like**
I'm a big fan of [direnv](https://direnv.net/) which allows me to have an environment based on which directory I'm in.
If the device was exposed as an environment variable (e.g. `TOIT_DEVICE`) I could keep that config local to a project directory, and not have to worry about deploying the right code to the right device

ptxmac, 2021-12-01:
Does this mean it's getting implemented?

jstroem, 2021-12-01:
Yes, its almost there.

closed 2022-06-30 by kasperl