toitware/public #43: [BUG] Toit CLI Installer for Windows v1.19.7

AlexanderPilhar, 2022-04-11:
**Describe the bug**
Latest installer v1.19.7 fails to execute. Installer v1.19.0 works fine.

**To reproduce**
1. Download installer
2. Execute installer
3. See error according to screenshot

**Expected behavior**
Installer should execute as expected.

**Screenshots**
[image]
NOTE: It doesn't matter in what directory the installer is placed.

**SDK and console version (please complete the following information):**
 - Version v1.19.7

**Additional context**
I still had version v1.19.0 of the installer stored on an other machine. I was able to install the Toit CLI v1.19.0 and update it via `toit update`.

floitsch, 2022-04-11:
Thanks for the report.
I tried to reproduce it, but have failed so far.
I tried a physical Windows 12 machine, and several virtual machines with different Windows versions. Toit installed without problems on all of them.

Could you please provide more information regarding your system?
- Which Windows version do you use?
- Do you have any Antivirus (or anti-malware) installed?
- anything else that could help us reproduce the issue?

Thanks

AlexanderPilhar, 2022-04-13:
I tried on two different machines, both running:
* Windows 10 (21H2, 19044.1586)
* Windows Defender

There might be some Group Policies causing this issue. I will try to install v1.19.7 on my private machine in the evening.

AlexanderPilhar, 2022-04-15:
Sorry, I've kept you waiting.
I managed to install Toit CLI for Windows v1.19.7 on my personal machine without any issues (Windows 11, 21H2, 22000.613).
Closing this now.

closed 2022-04-15 by AlexanderPilhar